umati/umatiGateway #122: Implement to use of a ca-cert store

GoetzGoerisch, 2025-04-09:
As a user of the umatiGateway I want to use a dedicated ca-root cert to use the gateway in scenarios, where I need to pass a TLS inspecting firewall.

Deployment:

OPC UA server -[1]-> umatiGateway -[2]-> TLS inspection FW -[3]-> Mqtt broker

connection 1 uses the selfsigned X.509 certificate of the umatiGateway
connection 2 uses a company CA-cert signed X.509 (TLS) certificate
connection 3 uses a Let's encrypt or Google CA signed X.509 (TLS) certificate

For the old gateway the following configuration options where used:

``` JSON5
    "CaCertPath":"", // path to the CA-Cert directory
    "CaTrustStorePath": "", // path to the CA-Cert file
```

Kantiran91, 2025-04-09:
@GoetzGoerisch Can we simulate this for testing in docker container?

GoetzGoerisch, 2025-04-09:
We need to asl @wlkrm what he used as a development env while developing this feature.

wlkrm, 2025-04-09:
I didn't have a test setup, it was a customer who requested this. I currently cannot remember who. But I think they use zscaler

GoetzGoerisch, 2025-04-10:
same customer now.

Kantiran91, 2025-04-14:
MQTTnet use the ceritficat manager of windows  maybe also linux

closed 2025-04-23 by mdornaus