umati/umatiGateway #144: MQTT connection issue

aschella, 2025-04-22:
When trying to connect to a local mqtt broker, there seem to be issues when resolvin the host name. 

This is either when the broker is not running on the standard port (1883) or not part of the docker compose file as described 
here. 
https://github.com/umati/umatiGateway/blob/develop/tests/integration/mqtt_test/docker-compose.yml

From the docker logs, I recheived the following error message: 

at UmatiGateway.OPC.MqttProvider.Connect_Client_Using_Tcp() in /src/umatiGateway/OPC/MqttProvider.cs:line 239
at UmatiGateway.OPC.MqttProvider.Connect(String connectionString, String connectionType, String port, String user, String pwd) in /src/umatiGateway/OPC/MqttProvider.cs:line 152
INFO Reconnecting Mqtt
ERROR Exception on Connecting to MqttBroker. MQTTnet.Exceptions.MqttCommunicationException: Error while connecting host 'Unspecified/localhost:1883'.

Kantiran91, 2025-04-23:
@GoetzGoerisch @mdornaus  Any ideas? why this makes problems?

Kantiran91, 2025-04-23:
@aschella It seems that the expose Port is only for the host. The container are in a local network. Can you change the port also in the mqtt config?

aschella, 2025-04-23:
> [@aschella](https://github.com/aschella) It seems that the expose Port is only for the host. The container are in a local network. Can you change the port also in the mqtt config?

Hi @Kantiran91 ,thank you for the advise. From the containers perspective, Gateway and broker communicate via the docker network, i. e. the mqtt broker via the port provided in the mosquitto.conf. 

Also, the gateway listens to the stndard port 4840 of the opc ua server, from that perspective. 

It seems to work now, I was able to receive messages: 

[image]

aschella, 2025-04-23:
I think this is not a bug, then.

closed 2025-04-23 by Kantiran91