umati/umatiGateway #152: No Payload for Published GMS Nodes

aschella, 2025-04-28:
I have started the Gateway, the umati server and the mqtt broker all within docker. Publishing of Information works for e. g. a "FullMachineTool". 

When declaring the root-node of the Basic GMS to be published in the corresponding Gateway Config:  

		<PublishedNode type="Numeric" namespaceurl="http://example.com/BasicGMS/" nodeId="66382"/>

I recheive an empty payload: 

[image]

Kantiran91, 2025-04-29:
The reason is that the namespace is not `"http://example.com/BasicGMS/"` - It is `http://www.isw.uni-stuttgart.de/BasicGMS/`

Kantiran91, 2025-04-29:
@mdornaus Maybe we should use the error topic again (Sending an Information that a configere Node is not found)

aschella, 2025-04-29:
Thank you for clarification. I was not aware of that. It works now. 

[image]

closed 2025-04-30 by aschella