umati/umatiGateway #179: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8)

AndreasHeine, 2025-06-10:
Hey @GoetzGoerisch,

i get:
`The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested`

```yaml
services:

  umatigateway:
    image: ghcr.io/umati/umatigateway:develop
    container_name: umatigateway
    ports:
      - 8080:8080
    expose:
      - 8080
    restart: unless-stopped

```

Have you run that on a Raspberry Pi at some point?

GoetzGoerisch, 2025-06-10:
https://github.com/umati/umatiGateway/blob/7e6af886c6169ea80d2eab6af0e3d04711047162/.github/workflows/build.yml#L146
We currently do not build arm64 images.

Kantiran91, 2025-06-13:
@AndreasHeine I think it works now?