umati/umatiGateway #208: Support for certificate-only authentication (no username/password) in OPC UA Client

Drixxen, 2025-07-11:
Dear umatiGateway Team,

as part of a research and integration task, I am currently working on connecting a coordinate measuring machine (CMM) that exposes its data via a standardized OPC UA information model. The goal is to translate the data into the umati PubSub structure for use in a dashboard or MQTT-based system.

However, the OPC UA server of the machine is only accessible via certificate-based authentication – that is, I have a client certificate and private key, but no username/password credentials. The connection works flawlessly using UAExpert, where I can connect by simply loading the certificate and key.

I would like to use the umatiGateway to connect to this server and publish the data, but I did not find any direct support or documentation for certificate-only authentication in the current gateway configuration or source code.

My questions:

Does the umatiGateway currently support client certificate authentication (without user/password)?

If not, would you recommend modifying the source code to load and attach the certificate manually within the OPC client logic (OpcUaClient.cs)?

Is there any roadmap or plan to support this natively?

Thank you in advance for your help. I’d appreciate any guidance or workaround to make this possible.

Best regards,
Hendrik (from Carl Zeiss AG)

Drixxen, 2025-07-11:
You can contact me: hendrik.lappe@zeiss.com

Kantiran91, 2025-07-11:
Thank you for the issues.
We currently work on different think and it is on the roadmap (see #154)
I try to upvote is in the next planings. 
Do you have some  deadlines?

Drixxen, 2025-07-11:
Thank you very much for your quick reply.

Yes, I do have a deadline – I need to have this running within the next month. I'm currently working on a technical paper and practical project which investigates how standardized OPC UA node structures can be translated and published in a scalable, interoperable way. Your gateway was recommended to me by an industry expert as the most promising tool for that.

Being able to use umatiGateway with certificate-based authentication (without username/password) would be a huge step forward for this work. It would allow me to demonstrate a real integration with a coordinate measuring machine (CMM) and show how standardized data can be streamed live into a broader digital infrastructure.

If there's any way to prioritize or temporarily unlock this feature, it would create massive value – not just for my project, but as a showcase of what’s already technically possible with OPC UA and MQTT in industrial environments.

In the meantime, do you happen to know of any other open-source tool or project that can handle this type of standardized mapping (e.g. OPC UA Machinery) and supports certificate-based authentication out of the box?
Even a simple converter or scriptable client would help to keep moving while waiting for this feature in umatiGateway.

Thanks again for your great work – and I’d be happy to test or validate any interim solution if that helps.

Best regards,

Drixxen, 2025-07-14:
Thank you again for your quick response – I really appreciate it.
If there’s anything I can do to support the implementation or testing of this feature, I’d be more than happy to help.

Best regards

Drixxen, 2025-07-14:
Thanks for the kind offer! Unfortunately, I only have access to a single internal server, which is part of a production monitoring process. I don’t have access or control to issue certificates for it. So to test things properly, I would either need to set up my own server or run the functionality locally on my machine.

mdornaus, 2025-08-01:
Ok I implemented it in the current dev verision. It currently only supports .pfx certificates but that should not be an issue cause you can generate .pfx out of the other certificate types by using openssl library.

There are two atttributes in OPC Connection certPath and CertPassword you can use to determine the certificate. If these fields are set they have precedence above user/password combination.

I attached a config file with the attributes:

<?xml version="1.0" encoding="utf-8"?>
<umatiGatewayConfig version="2.0" logLevel="Debug">
	<StartConfiguration startWebUI="True" startOPCConnection="False" startMqttProvider="False" startPubSubProvider="False"/>
	<WebUI url="https://127.0.0.1:7079"></WebUI>
	
	<OPCConnection serverendpoint="opc.tcp://localhost:4840" authentication="None" user ="" password="" ReadExtraLibs="False" certPath="mycert.pfx" certPassword="certPwd"/>
	<MqttProvider serverendpoint="mqtt://localhost:1883" user="company/client" password="" clientId="company/client" prefix="umati/v2" includeStructuredComponents="False" upperCaseRange="False" publishInterval="5000">
		<PublishedNodes>
			
			<PublishedNode type="Numeric" namespaceurl="http://example.com/BasicMachineTool/" nodeId="66382" baseType="" />
			
		</PublishedNodes>
		<CustomEncodings>
			<CustomEncoding name="GMSResultDataTypeEncoding" active="False" />
			<CustomEncoding name="ProcessingCategoryDataTypeEncoding" active="False" />
		</CustomEncodings>
		<IgnoredPlaceholderTags>
			<IgnoredPlaceholderTag name="&lt;ParameterIdentifier&gt;"/>
		</IgnoredPlaceholderTags>
	</MqttProvider>
	<PubSubProvider serverendpoint="mqtt://localhost:1883" user="" password="" clientId="company/client" prefix="umati/v3" allowUntrustedCertificates="false" publishInterval="10000" metaDataUpdateTime="10000">
		<PublishedNodes>
			
			<PublishedNode type="Numeric" namespaceurl="http://example.com/BasicMachineTool/" nodeId="66382" baseType="" />
			
		</PublishedNodes>
	</PubSubProvider>
</umatiGatewayConfig>

@Drixxen Maybe you can take a try with your machine. And give feedback.